n0m1/Sleep_n0m1 #3: Could you add compatibility to Due and Zero?

q2dg, 2015-10-02:
Thanks!

nullboundary, 2015-10-04:
Hi,

I don't own either the Due or the Zero so I have no way to test these. Also they are based on ARM micro-controllers  instead of AVR like the other Arduinos so the code in the library would be quite different. 

I do welcome pull requests if you can write it. 

If I get a my hands on a Due or Zero at some point I will definitely look into it.

nullboundary, 2015-10-07:
Sure, if you are going to write it there is lots of info here:

http://www.atmel.com/Images/Atmel-42412-Understanding-Performance-Levels-and-Power-Domains-AT04296_Application-Note.pdf

and in the datasheet starting on pages 190, and register info on page 209. 

http://www.atmel.com/Images/Atmel-42385-SAM-L21_Datasheet.pdf

narendradehury, 2016-06-06:
Hi, does it work with Arduino Mega. I tried it, but Arduino never woke up in POWER_DOWN mode with an interrupt from the DS3232 Alarm flag.

nullboundary, 2016-06-07:
It could work with the mega. However there is a small bug in the sleepInterrupt function preventing it from working. Its easy to fix though. I just haven't had a chance.

```cpp

void Sleep::sleepInterrupt(int interrupt,int mode) {

	if(mode == FALLING || mode == LOW)
	{
	   int pin = interrupt + 2; //will fail on the mega	
	   pinMode (pin, INPUT);
	   digitalWrite (pin, HIGH);
	}

	set_sleep_mode(sleepMode_);
	sleep_enable();
	attachInterrupt(interrupt,sleepHandler,mode);
	sei(); //make sure interrupts are on!
	sleep_mode();
	 //----------------------------- ZZZZZZ sleeping here----------------------
	sleep_disable(); //disable sleep, awake now
	detachInterrupt(interrupt);
}

```

As you can see the way its translating from interrupt number (0,1 etc) to the pin number is just adding 2. This worked for the Uno when the library was written and the mega didn't exist yet. However since the mega has more interrupts at higher pins it fails. It should work for the interrupts 0 (pin 2) and 1 (pin 3) on the mega. Just the higher interrupt pins will fail.

nullboundary, 2016-06-07:
Hi narendradehury,

I updated the function attachInterrupt in master. It now takes an interrupt pin number instead of the interrupt number and should work with the Mega.

```cpp
sleepInterrupt(int interruptPin,int mode)
```

Let me know if it works and I will update the release.

narendradehury, 2016-06-08:
Thank you Noah Shibley. Yeah it seems to have no issue with it. But in Arduino forum it was written all the sleep mode except SleepModeIdle it is mandatory to use interrupt LOW, but you have written LOW or FALLING.  Is it not going to put someone in trouble. Thnx anyway

nullboundary, 2016-06-10:
Ah good point. I will change it. 

So you can verify that its now working with the Mega? I will change the readme to say "Tested" for the Mega.

narendradehury, 2016-06-10:
Sure I will and let you know, when I am done. Cheers.

narendradehury, 2016-06-20:
yeah, it is good to go with Mega now. Tested.

nullboundary, 2016-06-22:
Awesome! Thanks!

closed 2016-06-22 by nullboundary