n0m1/Sleep_n0m1 #4: FALLING/LOW not working on bare 328

eried, 2015-12-21:
Does FALLING/LOW requires an external oscillator?

How can I wait for a pin to go LOW using the lowest energy? 

IF I use:
    while (!digitalRead(SENSOR_PIN));{
    sleep.pwrDownMode(); //set sleep mode
    sleep.sleepDelay(60000);}

It uses about 4-5 mA, but:
    sleep.sleepInterrupt(digitalPinToInterrupt(SENSOR_PIN), RISING);

uses <200 uA ... too bad

    sleep.sleepInterrupt(digitalPinToInterrupt(SENSOR_PIN), FALLING);
seems to sleep the atmega forever :/

nullboundary, 2016-02-23:
Hi, Did you figure out the problem? I am pretty sure LOW should work as well as HIGH without the external oscillator. 

Maybe there is a problem with the pin change interrupt? 

instead of this:
`
sleep.sleepInterrupt(digitalPinToInterrupt(SENSOR_PIN), FALLING);
`

try:
`
sleep.sleepInterrupt(0, FALLING); //number 0 (for digital pin 2) or number 1 (for digital pin 3) 
`

Another thing to try is using LOW instead of FALLING. 

Let me know if you can solve this. Thanks!

eried, 2016-02-29:
Hey thanks, this is related with the other issue about the calibration. The first calibration was happening = big consumption.

nullboundary, 2016-09-21:
#12 # #9 related to these other issues of calibration and edge vs level interrupts

closed 2016-09-21 by nullboundary