n0m1/Sleep_n0m1 #7: Working devices?

Andy-Voigt, 2016-04-10:
Hi,

can you please add a list of Arduino devices where you can truly say that the library works? A quick look at the code say "Fail on the mega" ?

Regards, Andy

nullboundary, 2016-04-11:
HI,

This library has only been tested on the Uno and Leonardo. The code says the way the interrupts are setup will probably fail on the mega. I think this should be easy to fix and I would welcome a pull request if you can debug and fix it. 

However if you are not using wake from sleep on interrupt then the library should work for the Mega. However I have never tested on the Mega (don't own one) so I don't really know. You can try it and let me know how it goes. 

I will add a comment about supported devices to the README, thanks for the tip!

nullboundary, 2016-04-11:
Actually a quick look at the code and the arduino interrupts reference tells me that the library should work with interrupts and the Mega. However the way the code in the library is implemented it will only work for interrupts 0 or 1. Which are pins 2 & 3 on the Mega. Since these pins are the same as the interrupt pins on the Uno. Other interrupts on the Mega will fail to work for this library as its currently implemented.

The sleepInterrupt() function should be re-writen to use digitalPinToInterrupt() and avoid these problems with the Mega.

maxholgasson, 2016-04-18:
Hi, I figured out that it works also great with a nano and a pro mini.  I have a device that measures temperature every hour however I'd like to interrupt sleeping when pressing a button for accessing a menu. How can I interrupt the Pro Mini?

nullboundary, 2016-04-19:
Hi Max,

The pro mini uses the same chip as the Uno, so the interrupts 0&1 are on the same pins, which are PD2 and PD3. The interrupt example should work for you. Let me know if there are any issues.

maxholgasson, 2016-04-19:
Hi, thanks for your reply.
I tried different ones but those didnt work. Can you give me a hint where to find the interrupt example ?

My device needs to wake up every 4 hours and if you press a button to enter a menu

nullboundary, 2016-04-19:
Hi Max,

Well there seems to be no way currently to do both sleepInterrupt and sleepDelay with the library, I see no reason why it couldn't be done by first attaching an interrupt then sleeping the device. Something like this untested code.

`
void loop() {
attachInterrupt(digitalPinToInterrupt(2),sleepHandler,CHANGE);
sleep.pwrDownMode(); //set sleep mode
interrupts(); //make sure interrupts are turned on!
sleep.sleepDelay(5000); //sleep for: sleepTime
}

void sleepHandler() {
//do something here.
}

`

maxholgasson, 2016-04-19:
Hi, I tried to combine your proposed code but it doens't wake up on interrupt
Did I put in wrong into the existing example?

`#include <Sleep_n0m1.h>

int count = 0;
Sleep sleep;
unsigned long sleepTime; //how long you want the arduino to sleep

int pin2 = 2;

void setup()
{
   
   Serial.begin(115200);
   pinMode(pin2, INPUT_PULLUP);
   sleepTime = 50000; //set sleep time in ms, max sleep time is 49.7 days
   
}

void loop()
{
  delay(100); //delay to allow serial output to be ready after wake up
  Serial.print("Awake for ");
  Serial.print(count);
  Serial.println("sec");

  count++;
  delay(1000);                           // waits for a second
  
  if(count >= 3)
  {
    count = 0;
    Serial.print("sleeping ");
    Serial.println(sleepTime); 
    delay(100); //delay to allow serial to fully print before sleep
    attachInterrupt(digitalPinToInterrupt(2),sleepHandler,CHANGE);
    sleep.pwrDownMode(); //set sleep mode
    interrupts(); //make sure interrupts are turned on!
    sleep.sleepDelay(sleepTime); //sleep for: sleepTime
  }
 
  
}`

nullboundary, 2016-04-19:
I don't see any problems but make sure you are creating a function to be called by the interrupt. The ISR (interrupt service routine).

```cpp
attachInterrupt(digitalPinToInterrupt(pin2),ISRFuncHere,LOW);

void ISRFuncHere() {
detachInterrupt(digitalPinToInterrupt(pin2));
}
```

however it could be that the interrupt needs to be LOW not CHANGE. Or perhaps somehow the Watch dog is causing issues with interrupt. Without testing this I am not 100% sure.

maxholgasson, 2016-04-19:
Mh, it doesn't work. I also tried FALLING  as interrupt but with the same result

[code]
#include <Sleep_n0m1.h>

int count = 0;
Sleep sleep;
unsigned long sleepTime; //how long you want the arduino to sleep
int pin2 = 2;

void setup()
{
 pinMode(pin2, INPUT_PULLUP);  
   Serial.begin(9600);
   sleepTime = 50000; //set sleep time in ms, max sleep time is 49.7 days
   
}

void loop()
{
  delay(100); //delay to allow serial output to be ready after wake up
  Serial.print("Awake for ");
  Serial.print(count);
  Serial.println("sec");

  count++;
  delay(1000);                           // waits for a second
  
  if(count >= 3)
  {
    count = 0;
    Serial.print("sleeping ");
    Serial.println(sleepTime); 
    delay(100); //delay to allow serial to fully print before sleep
    attachInterrupt(digitalPinToInterrupt(pin2),ISRFuncHere,LOW);
    sleep.pwrDownMode(); //set sleep mode
    interrupts(); //make sure interrupts are turned on!
    sleep.sleepDelay(sleepTime); //sleep for: sleepTime
  }
}

void ISRFuncHere() {
detachInterrupt(digitalPinToInterrupt(pin2));
} [/code]

nullboundary, 2016-04-19:
Does it just not wake up? 

whats up with the empty include? Just curious. I will test it out sometime maybe later this week if I get a chance. 

```
#include
```

maxholgasson, 2016-04-19:
oh, it's `#include <Sleep_n0m1.h>`

yes it doesn't wake up on external interrupt on pin2. The watchdog interrupt works

feilimb, 2016-04-23:
hi nullboundary, nice library many thanks.  do you happen to know whether the library supports any of the ESP-8266 chipset variants?

nullboundary, 2016-04-24:
Hi Feilimb,

Currently the library doesn't support the ESP-8266 and I am not sure that it is necessary. The ESP8266-arduino SDK has sleep built in that looks very similar to this library:

https://github.com/esp8266/Arduino/blob/master/doc/libraries.md#esp-specific-apis 

```
ESP.deepSleep(microseconds, mode);
```
 Will put the chip into deep sleep. mode is one of WAKE_RF_DEFAULT, WAKE_RFCAL, WAKE_NO_RFCAL, WAKE_RF_DISABLED. (GPIO16 needs to be tied to RST to wake from deepSleep.)

The only reason you may want it built into this library is so you could write the same code for multiple platforms. Is there a particular reason you would need this? I do have a bunch of nodemcus around and could write this but I didn't see any reason to do so.

feilimb, 2016-04-25:
Many thanks for the reply and info, my bad for not knowing the ESP library supported the different sleep modes.  Kudos for the offer of looking into support / porting but as you say it looks like the ESP library is more than sufficient in its current form.

closed 2016-04-27 by nullboundary