n0m1/Sleep_n0m1 #10: Atmega168 - error of compiling

SergioUno, 2016-08-28:
Arduino: 1.6.11 (Windows 7), Board: "Arduino Nano, ATmega168"

In file included from C:\...\Arduino\libraries\Sleep_n0m1examples\Sleep_n0m1_simple\Sleep_n0m1_simple.ino:1:0:

C:\...\Arduino\libraries\Sleep_n0m1/Sleep_n0m1.h: In member function 'void Sleep::extStandbyMode()':

C:\...\Arduino\libraries\Sleep_n0m1/Sleep_n0m1.h:87:37: error: 'SLEEP_MODE_EXT_STANDBY' was not declared in this scope

  void extStandbyMode(){setSleepMode(SLEEP_MODE_EXT_STANDBY);}

                                     ^

exit status 1
Error compiling for board Arduino Nano.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

nullboundary, 2016-09-07:
Seems like something has changed in the Arduino code base. Probably they updated avrlib which has broken this code. I will look into it.

nullboundary, 2016-09-21:
Looks like there were some changes to the avrlibc in some of the newer versions of Arduino IDE. With these changes SLEEP_MODE_EXT_STANDBY is no longer available for the ATmega168. 

[Check out this thread for more details](https://github.com/rocketscream/Low-Power/issues/14)

So your options are. 

1. Use and older version of the Arduino IDE (1.0.x)
2. Use a chip with the ATmega328
3. Use a different sleep mode. 

I hope that helps.

closed 2016-12-02 by nullboundary

Cubox-, 2018-08-22:
The compile error still has this issue, even when NOT using SLEEP_MODE_EXT_STANDBY.

Commenting out the void extStandbyMode(){setSleepMode(SLEEP_MODE_EXT_STANDBY);} line makes the compilation work fine.

There could be conditionnal #IF around this line, if the atmega is a 168.