n0m1/Sleep_n0m1 #11: Error on verifying code

lownote, 2016-09-02:
I get this error when I verify my code:

> C:\Users\Doug\Documents\Arduino\libraries\Sleep_n0m1-master\Sleep_n0m1.cpp: In function 'void sleepHandler()':

> C:\Users\Doug\Documents\Arduino\libraries\Sleep_n0m1-master\Sleep_n0m1.cpp:250:6:

>  warning: 'sleepHandler' appears to be a misspelled signal handler, missing __vector prefix [-Wmisspelled-isr]

>  void sleepHandler(void)

>       ^

I'm very new at this, so I apologize if there's something very simple I can do to fix this. I have no idea what this all even means, so any help would be appreciated. Below I'm including the code for my project, which is a laser pointer cat toy designed to turn on every hour to give the kitties something to do while I'm at work. 

```
#include <Servo.h>
#include <Sleep_n0m1.h>

Sleep sleep;
unsigned long sleepTime = 3600000;

Servo tiltservo;
Servo panservo;
int tiltpos = 0;
int panpos = 0;
int wait = 750;
int panmax = 105;
int panmin = 75;
int tiltmax = 105;
int tiltmin = 75;
int x = 0;
const int tilt = 3;
const int pan = 5;
const int lcd = 7;

void setup() {
  tiltservo.attach(tilt);
  panservo.attach(pan);
  pinMode(lcd, OUTPUT);
}

void loop() {
  tease();
  rest();
}

void tease(){
  digitalWrite(lcd, HIGH);

  for( int x=25; x>=0; x--){
      int scenario = random(0,2);
      switch(scenario){
    case 0:
    buzz();
    break;
    case 1:
    glide();
    break;
    case 2:
    morse();
    break;  
    }
  }
  tiltservo.write(tiltmin);
  panservo.write(panmin);
  digitalWrite(lcd, LOW);
  delay(2000);
}
void buzz(){
  for(int x=5; x>=0; x--){
    tiltpos = random(tiltmin, tiltmax);
    panpos = random(panmin, panmax);
    tiltservo.write(tiltpos);
    panservo.write(panpos);
    delay(500);
    for(int y=10; y>=0; y--){
      tiltpos = tiltpos + random(-4,4);
      panpos = panpos + random(-4,4);
      tiltservo.write(tiltpos);
      panservo.write(panpos);
      delay(500);
    }    
  }  
  delay(1500);
}

void glide(){
  tiltpos = tiltmin;
  panpos= panmin;
  int z = tiltmax - tiltmin;
  tiltservo.write(tiltpos);
  panservo.write(panpos);
  delay(50);
  for(x = z; x>=0; x--){
    tiltpos = tiltpos + 1;
    panpos = panpos +1;
     tiltservo.write(tiltpos);
     panservo.write(panpos);
     delay(50); 
  }  
  delay(2000);
}

void morse(){
  tiltpos = (tiltmin + tiltmax)/2;
  panpos = (panmin + panmax)/2;
   tiltservo.write(tiltpos);
   panservo.write(panpos);
   for(int x=10; x>=0; x--){
    digitalWrite(lcd, LOW);
    delay(500);
    digitalWrite(lcd, HIGH);
    delay(500);
   } 
   delay(2000);
}

void rest(){
  digitalWrite(lcd, LOW);
  tiltservo.write(tiltmin);
  panservo.write(panmin);
  sleep.pwrDownMode(); //set sleep mode
  sleep.sleepDelay(sleepTime); //sleep for: sleepTime
}

```

nullboundary, 2016-09-07:
Hmmm, seems like the same as the other issue. Probably the lastest version of arduino IDE has changed their version of avrlib and it broke this library. Your code looks fine.

lownote, 2016-09-07:
Thanks. 

I made the error go away by commenting out the second to last line to set sleep mode, but I'm not sure if its now actually going to sleep. 

I'm sure reverting back to the previous IDE will be fine for my purposes. Thanks!

nullboundary, 2016-09-09:
Yeah, without that last line its not actually sleeping.  Hopefully I can update the library soon.

nullboundary, 2016-09-21:
Hi, so I tested the library in the latest version of the Arduino IDE. Everything works fine, however I get the same error. Turns out if you read this carefully its just a warning not actually an error. So it can probably be ignored. Technically it should probably be fixed but its not affecting the usage of the library. So just use your original code and it should be sleeping just fine.

closed 2016-09-21 by nullboundary

lownote, 2016-09-21:
I think I've noticed this too while messing with the code more. 

Thanks. I'm new to this, so all of those compiler messages look really loud and alarming. I appreciate this tool!

borjaevo, 2018-06-29:
I have same problem here. I've checked with older arduino IDE version and not working also.

regisaabdalla, 2018-07-27:
We need to do  just a minor adjstment. 
In the "Sleep_n0m1.h" file, we  read:  _extern "C" void sleepHandler(void) __ __attribute__ ((**signal**));_

Just edit the file, to this: _extern "C" void sleepHandler(void) __ __attribute__ (());_

Hope to help!

Regards,
Regis

R0B-M4C, 2024-12-05:
Thank you, regisaabdalla!
I tried a few times here and managed to resolve the issue with the following change:

extern "C" void sleepHandler(void) __attribute__ (());

R0B-M4C, 2024-12-05:
Thank you, regisaabdalla!
I tried a few times here and managed to resolve the issue with the following change:

[image]