n0m1/Sleep_n0m1 #12: Does not work for pin High

bwheatl, 2016-09-13:
The code works great except when I monitor for a pin high condition. I changed falling and low to rising and high in the library. Any thoughts?

bwheatl, 2016-09-13:
I found all sleep modes except for idle requite pin LOW. Idle is the only one that allows pin HIGH and CHANGE.

nullboundary, 2016-09-21:
[image]

If you check out this chart from the Atmega328 datasheet you will see that in all sleep modes besides Idle only a level interrupt works (INT1,INT0 column). Level interrupts are only HIGH or LOW. Edge interrupts are CHANGE, RISING, and FALLING. 

Also the [Arduino reference](https://www.arduino.cc/en/Reference/AttachInterrupt) says only the Arduino Due, Zero, MKR1000 work for interrupt HIGH. 

So that leaves us with only LOW interrupt available for sleep modes besides Idle.

nullboundary, 2016-09-21:
This is interesting, found it [here](http://gammon.com.au/interrupts) But your saying this is not true? Maybe it depends on the version/batch of the Atmega chip? Strange. 

---

_Tip:
I received this confirmation from Atmel that the datasheet is wrong about only LOW interrupts waking the processor.

Quote:

Commented by Manoraj Gnanadhas (Atmel)
2015-01-20 06:23:36 GMT
[Recipients: Nick Gammon]

Hello Nick,

Our design team has confirmed that “Note-3 mentioned under Table 10-1” is a datasheet bug. So you can use any type of interrupt (Rising edge/ Falling edge / Low level / Any logical change) to wake up from sleep mode. Sorry for the inconvenience caused.

Best Regards,
Manoraj Gnanadhas

Thus, all interrupt types will wake the processor._

nullboundary, 2018-07-17:
Closing, hopefully its been solved or my answer helped.

closed 2018-07-17 by nullboundary