n0m1/Sleep_n0m1 #14: Sigfox after sleep

franckCordu, 2016-11-23:
Hello, thanks a lot for your library !

I'm trying to use it and I found a little problem. Maybe you can find the solutions quickly than me. 

I use an Arduino Uno with A sigfox modem to send data with a keypad. I use a button to wake my Uno with an interruption on pin 2. I send many datas in the first step of the sketch but after the sleep mode it's impossible to send anything. I have errors with send function. It's like the modem is not reachable. Like the connection is lost forever. A reset of the Arduino is not enough to make it works again. I have to unplug an re-plus the USB. 

Have you any idea where the problem come from ?

franckCordu, 2016-11-24:
I found the solutions. Its a conflict between serial.print and sigfox. Without serial.print everything works fine. Thanks

closed 2016-11-24 by franckCordu

nullboundary, 2016-12-02:
Sorry for the slow reply, Glade you could figure it out. Yeah serial.print is a common source of bugs, one of the first things I check while debugging mysterious issues. Somewhere I saw a library that logs the serial.printlns to flash or eeprom and then prints them later at a time when the processor is not busy, to avoid these kinds of issues. I think its all about processor cycles and timing. Another thing that sometimes helps is increasing the baudrate of the serial. Since then sending the data will take less time.