n0m1/Sleep_n0m1 #15: how to wake from  sleep.sleepDelay(600000);  through an interrupt ?

DimMedvedev, 2016-12-24:
help me please
how to wake from  sleep.sleepDelay(600000);  through an interrupt ?

nullboundary, 2017-01-05:
Hi DimMedvedev,

The library is currently not really designed to do that. It is however a good idea for an added feature.  You could make such a  function by re-writing sleepPinInterrupt to set the interrupt, then call sleepDelay inside that function. It could be called sleepDelayOrInterrupt() ?

DimMedvedev, 2017-01-05:
I came up with this method:
`#include <Sleep_n0m1.h>

Sleep sleep;
boolean abortSleep; //cancel sleep cycle
unsigned long sleepTime; //how long you want the arduino to sleep

void setup()
{
   
   Serial.begin(9600);
   pinMode(2, INPUT);   
   abortSleep = false; //can be used to cancel the sleep cycle
   sleepTime = 50000; //set sleep time in ms, max sleep time is 49.7 days

}

void knopka()
{
  abortSleep = true;
}

void loop()
{
  
  delay(100); ////delays are just for serial print, without serial they can be removed
  Serial.println("execute your code here");

  Serial.print("sleeping for ");
  Serial.println(sleepTime); 
  delay(100); //delay to allow serial to fully print before sleep

  attachInterrupt(0, knopka, CHANGE );
  sleep.pwrDownMode();                                //set sleep mode
  sleep.sleepDelay(sleepTime,abortSleep);            //sleep for: sleepTime
  detachInterrupt(0);
  abortSleep = false;  
  
}
`

DimMedvedev, 2017-01-05:
but my code is not perfect

closed 2017-01-05 by DimMedvedev

nullboundary, 2017-01-05:
Does it work? Without testing it I wouldn't know. Looks like it might work. I was thinking more about combining the two function in the library. 

```
void Sleep::sleepDelay(unsigned long sleepTime, boolean &abortCycle, int interrupt, int mode) {

int intNum = digitalPinToInterrupt(interruptPin);

	if(mode == FALLING || mode == LOW)
	{
	   pinMode (interruptPin, INPUT);
	   digitalWrite (interruptPin, HIGH);
	}

  ADCSRA &= ~(1<<ADEN);  // adc off
   // PRR = 0xEF; // modules off

  ++sleepCycleCount;
  sleepCycleCount = sleepCycleCount % sleepCycleInterval; //recalibrate every interval cycles
  if(sleepCycleCount == 1)
  {
	calibrateTime(sleepTime,abortCycle);
  }
  else
  {
  	set_sleep_mode(sleepMode_);
       
       //add these two lines
       attachInterrupt(interrupt,sleepHandler,mode); 
	sei(); //make sure interrupts are on!
  	
       int trem = sleepWDT(sleepTime*calibv,abortCycle);
       
       /add this line. 
       detachInterrupt(interrupt);
  	timeSleep += (sleepTime-trem);
  }
  // PRR = 0x00; //modules on
 ADCSRA |= (1<<ADEN);  // adc on
}
```

Maybe something like this. Its just a guess though.

veer5551, 2018-07-17:
Hello!
I need help regarding the same issue. Device should wakeup from interrupt as well as sleep at the same time. Any help or update regarding the same is greatly appreciated.
Thank you.