n0m1/Sleep_n0m1 #16: Using interrupt with sleepDelay in pwrDownMode

divya0056, 2017-01-25:
Hello, trying to keep Arduino in sleep mode until one of the two events have occurred 
1 if the desired time has passed , Arduino wakes up does its thing. OR if the time has NOT  passed but interrupt pin becomes active Arduino wakes up. 
Is this possible?
Thanks for you help.

divya0056, 2017-01-26:
Found closed issue by DimMedvedev and your input. Modified the .CPP file with your input , Having issues to get the code complied. Could you please make changes to the library. I can test it and let you know  .  Thanks.

divya0056, 2017-01-28:
Ok, was able to compile modified .cpp and .h files with code from n0m1. 
When abortCycle is true, arduino stays in sleep until interrupt  button is pressed. But it does not wake up by itself if the "sleepTime " has passed. 
When abortCycle is false, arduino does not wake up even if the interrupt button is pressed until sleepTime  has passed ( mandatory sleep). After sleepTime has passed arduino will wake up only if interrupt  button is pressed.

My goal : if interrupt  button is pressed regardless of time passed wakeup. if no interrupt  button is pressed wake up if time has passed. 

/************************************************************************************
 *
 * 	Name    : Sleep_n0m1.h
 * 	Author  : Noah Shibley / NoMi Design
 * 	Date    : July 10th 2011
 * 	Version : 0.1
 * 	Notes   : Some of this code comes from "Cloudy" on the arduino forum
 *			  http://www.arduino.cc/cgi-bin/yabb2/YaBB.pl?num=1292898715
 *
 * 		    Sleep_n0m1 is free software: you can redistribute it and/or modify
 * 		    it under the terms of the GNU General Public License as published by
 * 		    the Free Software Foundation, either version 3 of the License, or
 * 		    (at your option) any later version.
 *
 * 		    Sleep_n0m1 is distributed in the hope that it will be useful,
 * 		    but WITHOUT ANY WARRANTY; without even the implied warranty of
 * 		    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * 		    GNU General Public License for more details.
 *
 * 		    You should have received a copy of the GNU General Public License
 * 		    along with Sleep_n0m1.  If not, see <http://www.gnu.org/licenses/>.
 *
 ***********************************************************************************/

#include "Sleep_n0m1.h"

Sleep* Sleep::pSleep = 0;

Sleep::Sleep()
{
	pSleep = this;	//the ptr points to this object
	timeSleep = 0;  // total time due to sleep
	calibv = 1.0; // ratio of real clock with WDT clock
	byte isrcalled = 0;  // WDT vector flag
	sleepCycleCount = 0;
	sleepCycleInterval = 100;

}

/********************************************************************
*
*	setSleepMode
*
********************************************************************/
void Sleep::setSleepMode(int mode)
{
  sleepMode_ = mode;
}

/********************************************************************
*
*	calibrateTime
*
********************************************************************/
void Sleep::calibrateTime(unsigned long sleepTime, boolean &abortCycle) {
  // timer0 continues to run in idle sleep mode
  set_sleep_mode(SLEEP_MODE_IDLE);
  long tt1=millis();
  sleepWDT(sleepTime,abortCycle);
  long tt2=millis();

  calibv = (float) sleepTime/(tt2-tt1);

  //Serial.println(calibv);
[... 378 lines cut ...]

void newSleep(unsigned long sleepTime, boolean &abortCycle, int interruptPin, int mode) ;

private:

  int sleepMode_;
  unsigned long timeSleep;
  float calibv;
  volatile byte isrcalled;
  static Sleep* pSleep; //static ptr to Sleep class for the ISR
  int sleepCycleCount;
  int sleepCycleInterval;

   void setSleepMode(int mode);
   void WDT_Off();
   void WDT_On(byte psMask);
   int sleepWDT(unsigned long remainTime,boolean &abortCycle);
   void calibrateTime(unsigned long sleepTime,boolean &abortCycle); //calibrate the time keeping difference between WDT and Timer0
   unsigned long WDTMillis(); // Estimated millis is real clock + calibrated sleep time

};

#endif

// any help greatly appreciated.

veer5551, 2018-07-17:
Hello!
I have the same issue. Any update or help regarding the same is greatly appreciated.
Thank you.

nullboundary, 2018-07-17:
Hmmm, as the library is now there isn't any simple way to do this. However the sleepPinInterrupt library function could be rewritten to pass in a function for the sleep interrupt handler. Something like this:

    void Sleep::sleepPinInterrupt(int interruptPin,int mode, std::function<void()> handler){
 
	int intNum = digitalPinToInterrupt(interruptPin);

	if(mode == FALLING || mode == LOW)
	{
	   pinMode (interruptPin, INPUT);
	   digitalWrite (interruptPin, HIGH);
	}

	set_sleep_mode(sleepMode_);
	sleep_enable();
	attachInterrupt(intNum,handler,mode);
	sei(); //make sure interrupts are on!
	sleep_mode();
	 //----------------------------- ZZZZZZ sleeping here----------------------
	sleep_disable(); //disable sleep, awake now
	detachInterrupt(intNum);
    }

Then define that handler function in your sketch before you pass it into the sleepPinInterrupt function. Or use a C++11 lambda function (similar to a anonymous function in javascript) like so:

    sleep.sleepPinInterrupt(intPin,LOW,[](){ 
        //do interrupt handler stuff here
    });

I haven't tested any of this stuff, but if you can make it work and make a pull request I'll be sure to add it in to the library.