n0m1/Sleep_n0m1 #19: Attaching two different interrupts

JMattChiam, 2020-12-01:


JMattChiam, 2020-12-01:
Hi, i'm trying to attach two different interrupts to wake up the arduino. One when a button is pressed, and another connected to an external timer. How would i go about doing this? 
Thanks!

nullboundary, 2020-12-01:
Hi, I don't know exactly as I have no way to currently test this, however I would think perhaps you could combine the examples Sleep_n0m1_Interrupt and Sleep_n0m1_abortsleep. You could use the external timer to set the abortsleep flag and set the interrupt pin to the button pressed. You need to make sure you set the sleep mode so that you don't sleep the timer you want to use as well. Try that and let me know how it goes!

JMattChiam, 2020-12-03:
Hi! 
I realised that just attaching an additional interrupt with attachInterrupt() is sufficient for my purposes (just change a flag that runs a function in the main loop before going back to sleep). My main timer is still attached to sleepPinInterrupt. 
Thanks for the help!

closed 2020-12-03 by JMattChiam