n0m1/Sleep_n0m1 #21: Sleep time not working correctly

alexandier2002, 2021-05-09:
Hello, i would like to ask something because i have a problem with your library (i guess). I need the arduino to run every 12 hours so on sleep time i write :sleepTime = 43200000; which is 12 hours but in reality it doesnt run every 12 hours but takes more time and its inaccurate. Any ideas what can be the problem? 
Thank you.

nullboundary, 2021-05-21:
Hi alexandier2002, 
 
You will probably have to provide more details. I can't say offhand what could be wrong. However I have usually used the library for shorter time cycles for more accuracy. Instead of sleeping for 12 hours, I would sleep for maybe 30 minutes, 24 times in a loop, then on the 24 time do whatever you want to do every 12 hours and then start everything over. The way the library is designed is it uses the first sleep cycle for calibration so with such a long sleep time cycle it might be off on the first cycle. Also the library uses the watchdog timer to wake itself up from sleep which can be quite sensitive to temperature variations, which is why the timer needs to be calibrated on the first loop and re calibrated every 100 sleep cycles. If your cycles are too long the watchdog timer might be tracking off time and not have a chance to re calibrate because its asleep. Let me know if this helps.

alexandier2002, 2021-05-21:
Hello noah it seems that what cause my problem was a loose wire from the power sypply. I know about the first cycle cause i read it from other post. In my case the first cyrcle send the next data in 13h but after that its back to 12h. It only skips some min but ofc as you said its based on the temperature. Thank you so much for your answer though it was really helpful.

closed 2021-05-24 by nullboundary